darklife/darkriscv #1: Useless commit messages

dolmen, 2018-08-29:
Commit messages have all "Changes to be committed:" as the first line and then contain only a list of filename. This is is quite useless (well, unless to write a big project in one night).

See best practices: https://chris.beams.io/posts/git-commit/

ddimmich, 2018-08-29:
Which bit of written in 1 night did you miss ;)

z3ntu, 2018-08-29:
@ddimmich As far as I understand, everything that was the 1 night is in the first ~3 commits as the 4th commit adds the "somewhere between 2am and 8am" in which it was developed to the README and there are many commits after that.

samsoniuk, 2018-09-17:
After decades away from the open source community, my lack of ability with this new tool called "git" is very evident! Thank you for point the problem and thank you for the best practices link! :)

closed 2018-10-14 by samsoniuk