darklife/darkriscv #4: Need help simulating with iverilog and gtkwave

mytbk, 2018-09-12:
I'm trying to use iverilog and gtkwave to see how this RISC-V core works. First, I modify the darksim.v to add the messing parameter of darksocv, and modify darksocv.v to dump core0 to a vcd file.

```
[irudog@mylaptop darkriscv]$ git diff
diff --git a/rtl/darksocv.v b/rtl/darksocv.v
index 88b5c83..310274e 100644
--- a/rtl/darksocv.v
+++ b/rtl/darksocv.v
@@ -32,7 +32,7 @@
 
 // pseudo-soc for testing purposes
 
-`define CACHE_CONTROLLER 1
+`define CACHE_CONTROLLER 0
 //`define STAGE3           1
 
 module darksocv
@@ -271,4 +271,9 @@ module darksocv
         .DEBUG(DEBUG)
     );
 
+       initial begin
+               $dumpfile("darkriscv.vcd");
+               $dumpvars(0, core0);
+       end
+
 endmodule
diff --git a/sim/darksimv.v b/sim/darksimv.v
index 5247e45..4fe8eb2 100644
--- a/sim/darksimv.v
+++ b/sim/darksimv.v
@@ -42,6 +42,9 @@ module darksimv;
 
     always@(posedge CLK) RES <= RES ? RES-1 : 0;
 
-    darksocv darksocv(CLK,|RES);
+       wire [31:0] dummyin = 32'b0;
+       wire [3:0] dummyout;
+
+    darksocv darksocv(CLK,|RES, dummyin, dummyin, dummyin[0], dummyin[0], dummyout);
 
 endmodule
```

But after I watch the wave in gtkwave, I saw a problem. When PC advances as expected, the instruction input IDATA is always xxxxxxx, I don't know why this happens.

Can you give a good instruction on how to simulate the darkriscv core using iverilog and gtkwave?

samsoniuk, 2018-09-17:
Wow! I made exactly the same changes as you and the iverilog was unable to locate the **darksocv.hex** file, but when I changed to the **sim** directory, everything worked fine:

**cd sim**
**iverilog -o darksim darksimv.v ../rtl/darksocv.v ../rtl/darkriscv.v**

The **darksim** file, when executed, produced the following result in the console:

**./darksim** 
_WARNING: ../rtl/darksocv.v:69: $readmemh(../src/darksocv.hex): Not enough words in the file for the requested range [0:511].
VCD info: dumpfile darkriscv.vcd opened for output.
cache controller active
hhhhello world!_

In order to stop, I typed a control+c and typed finish in the iverilog console. Of course, the dump file is increasing in size as long the processor is looped at end of the hello world code. The warning about the the **darksocv.hex** length appears to not be a problem, as long the file is smaller than the reserved space in the darksocv.

In order to limit the dump file, I included an additional commando in order to finish the execution after 10us in the **darksimv.v**:

**initial #(10e3) $finish;**

And the simulation generated a smaller dump file with 177KB. Of course, as long the hello world was displayed in the console, we can infer that the simulation worked as expected! Of course, the cache was not really disabled, but the iverilog simulated it in the correct way! :)

Anyway,I loaded the resulting **darkriscv.vcd** in the GTKWave and I found the hierarchy with the darksimv, darksocv and the darkriscv. When the signals are displayed, they are scaled in a way only the first ps are displayed and everything is undefined. Of course, the logic need some clocks in the RESET state in order to get defined results! 

As long the simulation is now limited to 10us, is possible use the Zoom Fit option in the GTKWave and everything is correctly displayed (the RESET will be released at ~226ns).

Well, I will made additional tests regarding the iverilog and I will try include a section about the iverilog in the README.md! :)

samsoniuk, 2018-10-14:
I made additional changes in order to support the Icarus by default, including a Makefile in the sim directory, in order is possible generate the VCD file directly!

closed 2018-10-14 by samsoniuk