darklife/darkriscv #7: rename darkrisc to norisk

gsilos, 2018-11-20:
New projects should not be using dark in its name. Using semantically more meaningful terms is better than terms that conjure up images of humans forced to work into dark places.

Suggested alternates:

norisk
riskless

samsoniuk, 2018-11-21:
Thank you by your suggestion conrade Silos! Although your observation is perfectly valid, you must remember that I am a *physicist* and the prefix dark- is very traditional in my study area, in special when we have only a vague idea how something works. By this way, as long the *darkriscv* is an experimental processor, the prefix dark- appears to be appropriate.

samsoniuk, 2018-11-21:
I hope this picture can help:

[image]

closed 2018-12-12 by samsoniuk