darklife/darkriscv #8: I think there is a mistake in the RTL code

hyf6661669, 2018-12-08:
In darkriscv.v, the following code is used to choose the data that will be written into the register file:
```
 REG[DPTR] <=   RES ? RESET_SP  :        // reset sp
                       HLT ? REG[DPTR] :        // halt
                     !DPTR ? 0 :                // x0 = 0, always!
                     AUIPC ? NXPC+SIMM :
                      JAL||
                      JALR ? NXPC :
                       LUI ? SIMM :
                       LCC ? LDATA :
                       MCC ? MDATA : 
                       RCC ? RDATA : 
                       CCC ? CDATA : 
                             REG[DPTR];
```
For instruction "AUIPC", the data should be PC + SIMM. But I only tested it by using the core as a 2-stage CPU without cache. I hope you can have a look.

samsoniuk, 2018-12-09:
Good point! I will check!

hyf6661669, 2018-12-11:
> Good point! I will check!

By the way, which simulation tools do you use for simulation? I am using ModelSim 10.4c, but the operator ```>>>``` for instruction ```SRA or SRAI``` doesn't work. It still performs "shift right logical".

samsoniuk, 2018-12-11:
I am testing with iSIM from Xilinx ISE 14.7, but maybe is possible write that same expression in a more compact and portable way.

closed 2018-12-24 by hyf6661669