darklife/darkriscv #11: Instruction for compiling on Arty board

komminenibalaji, 2019-01-14:
Hi Samsoniuk,

Sorry this is not an issue but couldn't figure another way to ask.

I was wondering if you could help with instructions on how to build this project for Arty board from digilent. 

Thanks,
Balaji

hyf6661669, 2019-01-17:
Actually you can look at darkriscv/ISE for reference, it is really simple. 

At present darkriscv is a risc-v core and seems not be able to control any peripherals. So you just need to use some Block RAMs in your Xilinx FPGA and load the test programs into them.

komminenibalaji, 2019-01-17:
Thanks Samsouniuk, I will try it out.

samsoniuk, 2019-02-12:
I am not sure it can help about the Vivado, but I added support for 3 different boards in the directory ise/boards: Avnet Microboard w/ Spartan-6 LX9, Xilinx AC701 w/ Artix-7 A200 (thanks to a Xilinx FAE) and QMTech SDRAM w/ Spartan-6 LX16. Basically, the difference between that boards are the pin definition and the core clock (which must be known in order to generate the correct 115200 bps baudrate for the UART). In some degree it is very funny that the Artix-7 A200 *is* supported by ISE, although I guess there is a possible performance limitation: the synthesis shows up to 147MHz, the the implementation barely reached 90MHz. Anyway, I will have short access to a Microzed board in some weeks and I guess will be possible try add a limited support for the Zynq and Vivado.

samsoniuk, 2020-01-05:
Vivado is finally supported, although is not possible yet automatize the build via a Makefile, but at least there is an example for Spartan-7 available here:

https://github.com/darklife/darkriscv/tree/master/boards/qmtech_spartan7_s15

Of course, although the changes appears to be small, other changes were done in the last weeks in order to make the simulation and synthesis work with Vivado.

closed 2020-01-05 by samsoniuk