darklife/darkriscv #12: RV64 possibilities?

drudru, 2019-02-15:
Hi,

What is your estimate of supporting 64 bit RISC-V. I'm still pretty new to this, Could it run 100mhz on an Artix board? Would it even fit?

samsoniuk, 2019-02-16:
Hi Dru,

In theory the support for RV64I is easily implementable and I can check this topic in this weekend. Of course, large data buses and additional instructions will probably impact the performance, in a way that the RV32I version will probably be faster than the RV64I version. Anyway, the performance will depends of the FPGA selected.

About the Artix-7, I tested the darkriscv in a Xilinx AC701 board equipped with an Artix-7 A200 and everything worked fine. The support for the AC701 is available in the directory ise/board/xilinx_ac701_a200, but unfortunately it is supported only in the Xilinx ISE, which supports a small number of FPGAs from the 7-series. In the case of a different Artix-7 FPGA, you will probably need import the ISE project to Vivado.

Well, the single core darkriscv w/ 4KB ROM, 4KB RAM and UART requires around 1400 LUTs, with is around 1% of an Artix-7 A200. I guess the smaller Artix-7 can easily support some darkriscv cores at same time. the performance is more complex: although the synthesis pointed that the darkriscv can work up to 147MHz in the Artix-7, for some reason the implementation steps barely reached the performance of 90MHz. 

Maybe the problem is related to the tool (I used a very old Xilinx ISE tool instead of the more recent Vivado), maybe the problem is related to the optimization features.

I will make some tests regarding the RV64I support and I will add an additional feedback about it later.

drudru, 2019-02-19:
BTW - numato makes a relatively inexpensive Artix-7 board:
https://numato.com/product/mimas-a7-artix-7-fpga-development-board-with-ddr-sdram-and-gigabit-ethernet

hyf6661669, 2019-02-21:
@drudru 
HAHA, actually there are lots of inexpensive Xilinx FPGA boards in Chinese biggest e-commerce sites, www.taobao.com.

samsoniuk, 2019-02-23:
I found a very inexpensive Artix-7 board equipped w/ SDRAM from QMTtech: 

https://www.aliexpress.com/item/QMTECH-Xilinx-FPGA-Artix7-Artix-7-Core-Board-XC7A35T-32MB-SDRAM/1000006687962.html

Of course, as other QMTech boards, it does NOT include JTAG, UART or GbE interfaces as you can found in the Numato boards. Anyway, I have two QMTech boards w/ Spartan-6 LX16 and they works fine with cheap JTAG and USB/UART adapters:

https://www.aliexpress.com/item/Xilinx-Platform-Cable-USB-FPGA-CPLD-Download-the-debugger-Support-the-JTAG-Slave-Serial-SPI-is/32805986808.html

https://www.aliexpress.com/item/1pcs-CP2102-module-USB-to-TTL-serial-UART-STC-download-cable-PL2303-Super-Brush-line-upgrade/32694152202.html

About the RV64I, I started adapt the support for RV32I/RV6I via ifdefs, but in the process introduced some bugs which makes the RV32I version crash and I will need more time in order to remove all bugs. In a preliminary way, there is no impact in the RV32I performance, but the RV64I requires more logic regarding the memory interface, since the memory interface logic must support all possible combinations of 8/16/32/64 bit operations in different alignments.

closed 2021-01-29 by drudru