darklife/darkriscv #13: Suggestion for higher frequency

hyf6661669, 2019-02-21:
In #12 , you said that on Artix-7 200T, the darkriscv can barely reach the performance of 90Mhz. I want to give some suggestion:

1. I have found that you use opposite clk edge for RAM/ROM and cpu, in the 2-stage configuration. That will slow down the speed because the "Load" instruction must get data from RAM and write it into the registers. I suggest that implement a 3-stage configuration without I/D cache. As much as I know, many low-end ARM cores, such as cortex M0, M1, M3, M4, don't have caches. They only have some embedded SRAM and FLASH. And they are designed for simple application scenarios, which don't need much room for user codes and data. Therefore cache might be uncessary and dozens of KByte of SRAM is enough.

[image]

2. Use the latest Vivado for synthesis and implementation instead of ISE 14.7. I find that the result of Vivado is better than ISE for Xilinx 7 series FPGA.

samsoniuk, 2019-02-23:
I am planning reintroduce the 3-stage pipeline version, but with extra modifications in order to not require a cache controller. As long the difference is basically in the pre-fetch, there is no much impact in the logic consumption. Basically, the solution is introduce one extra wait-state for all load instructions, in order to meet the blockram requirement. With a single edge, I think the clock will increase ~20%, which will be compensated by a 3-clock branch (1 clock for the branch and 2 to flush the pipeline) and a 2-clock load (one clock to read from the blockram and another to write to the destination register).

samsoniuk, 2019-02-23:
Another planned change regards to the shift operations. In some tests I commented only the shift operations and I found that the shift operations are probably responsible by around 20-25% of the logic, which probably impacts in the clock performance because makes the path from register to register too long. Basically, the shift logic to shift a register by 32 bits right or left in a single requires lots of logic. The smaller and faster solution, from the logic point of view, is use a clocked shift register, which may requires 0-32 clocks to shift the register in any direction. The best solution will probably be anything mixing the clocked shift register and additional logic, for example, a smaller switch in order to shift by n/8 and a clocked shift register in order to shift the remaining n%8 shifts.

hyf6661669, 2019-02-24:
@samsoniuk 
Very good idea!

closed 2019-02-24 by hyf6661669

daybyter, 2019-05-29:
You could implement shifts for 16,8,4,2,1 bits and just use the bits of the shift value to activate those shift operations. So you can shift up to 31 bits in max 5 cycles.