darklife/darkriscv #15: Adding a In-System Programmability(ISP) function?

hyf6661669, 2019-03-11:
I advise adding a ISP function to darkriscv. At present, if I write some new C codes and generate new data for ram and rom, I have to use ISE or Vivado to run synthesis and implementation again, which takes lots of time . 

If PC can send these data to BlockRAMs for ram and rom via USB/JTAG/UART interface, it would be very nice. So the ```.mcs``` file can only store information about darkriscv itself, which means less time for running ISE or Vivado. We can also use some buttons to control this progress.

hyf6661669, 2019-03-13:
I found "updatemem" function of Vivado could be used to initialize BRAMs without rerun the synthesis and implementation, but it is hard to understand. I will try to find out how to use it.

samsoniuk, 2019-05-19:
Thanks for you suggestion HYF! I am currently away from the darkriscv due to other tasks, but I hope I will return in some weeks... About the ISP feature, although you are investigating about the updatemem, my plan is implement in the future a small boot firmware in order to provide support for software upload via UART. This solution is not so complex and may enable program the darkriscv without rebuild the FPGA.

samsoniuk, 2020-10-22:
There are at least three possible ways to handle the problem regarding the ISP feature:

a) a small ROM w/ support to boot from UART or from SPI FLASH
b) a double-port ROM that can be loaded by a supervisor core, that have a separate ROM w/ a debug code, in a way that is possible freeze the user thread, manipulate the user memory, read/write the user thread registers (via CSR instructions), set breakpoints, etc
c) a dual-thread core that have a separate ROM w/ a debug code in the supervisor thread, like the option b), but in the same core.

The first option appears to be the easiest by far, the other two appears to be good too...

closed 2023-10-13 by samsoniuk