darklife/darkriscv #18: Be confusing about interrupt

thomascp, 2019-07-22:
Hi Marcelo,

I have read these new codes which are commited recently.
I am confusing about NXPC2[2] thing, which index 0 and 1 represent different threads if I am right.
And in C codes, 'int tmp = 1&threads++;' is used to distuguish different threads.
My question is why to implement these two PC values?
In my opinion, if an interrupt happens, the PC register should be configured to a special value, in RISCV it is mtvec, and PC should be reverted back the old value if the interrupt returns, right?
It is hardware who decides whether to go into interrupts, and it is OS who decides whether to change the current thread to another one and save some current context.
So why not to implement mtvec and something related to that? 
At least, for your new codes, I couldn't understand your purpose, and there is even no save context operations for thread switching.
So please tell me your purpose and thoughts about this if you are free, thanks!
And if you have any email or slack, please let me know if you are OK with it.

BR,
Peng

samsoniuk, 2019-07-25:
Hi Peng,

Thank you for you feedback! Although the macro regarding this new code is named \_\_INTERRUPT\_\_, the real purpose of this new code is test the SMT concept (simultaneous multi-threading), which means that I will probably rename the macro to \_\_SMT\_\_ in the future, in a way that will be possible set an arbitrary number of threads between 1 and n. Although my current applications for the darkriscv are single-thread applications around a event loop, I will need support between 8 and 16 preemptive threads in the future and I think the SMT concept will work well for my future applications.

In order to explaining how SMT concept works in the darkriscv, we need start with the traditional concept, which use the following registers:

PC = current program counter
NXPC = program counter + 1 instruction
NXPC2 = program counter + 2 instructions
REG1[0:31] = array of 32 registers regarding the S1PTR register
REG2[0:31] = array of 32 registers regarding the S2PTR register

The use of REG1 and REG2 regards that the some RV32I instructions requires 3 operands: two source registers and one destination register. Although the synthesis tool can easily optimize this by duplicating the resource automatically, I found that the performance appears to be better when the resource is duplicated by hand and the paths are explicit defined. In addition, PC can be deterministically updated by the value of NXPC delayed by one clock and NXPC can be deterministically updated by the value of NXPC2 delayed by one clock. There are other internal registers, but they works more or less in the same way as the PC and NXPC, in a way that they are not really relevant (i.e. they will be updated by something else delayed by one or more clocks).

This means that, when expanding to a SMT concept in order to support *n* threads, we will need expand the following registers in order to preserve the thread context:

NXPC2[0:n-1] = n\*(program counter + 2 instructions)
REG1[0:32\*n-1] = array of n\*32 registers regarding the S1PTR register
REG2[0:32\*n-1] = array of n\*32 registers regarding the S2PTR register

In the case of the \_\_INTERRUPT\_\_ support, I just defined n = 2, in a way that the test works with two separate PCs and two separate sets of 32 registers, resulting in two fully independent contexts. The separation of NXPC2[] in two different contexts is natural, since it is indexed by XMODE, but the separation of REG1[] and REG2[] in two different contexts is done via the { XMODE, S1PTR } and { XMODE, S2PTR } indexes, in a way that the registers 0-31 corresponds to the thread 0 and the registers 32-63 to the thread 1.

However, there are two problems: *when* swap the threads and *how* initialize each thread context at the core startup... of course, *every time* the core is reset and not only after the FPGA is loaded.

The problem regarding *when* swap the threads is solved part by a periodic timer, which defines the start of the swap window (and the period of the window), and in part by the pipeline flush logic activated by taken branches, which defines the end of the swap window. The use of the pipeline flush activated by taken branches enables shadow the time required to flush the pipeline in the thread swap with the pipeline flush already required by the taken branch, i.e. the thread swap is done in the idle time after a taken branch.

Also, as bonus, this approach ensures that the code runs continuously in an atomic way until the next taken branch, solving the second problem (how initialize each thread context). In this case, all threads just start with the same PC and run the same code, but as long the code works in an atomic way, it is possible that each thread runs the entire sequence of code "tmp = 1&threads++" in an atomic way, resulting in different values for *tmp* when different threads run the code, which result in different program path decisions, in a way that the thread 0 will jump to the main() function and the thread 1 will run the interrupt loop.

There is only an open issue with this scheme: both threads use the *same* stack area. Although there are some risks here, the code is designed in a way that the thread 1 does not write anything in the stack... but, of course, something more complex will require some assembly in order to set different stack areas for the different threads. This means that this is not really an issue, as long you known how handle it.

Another possibility is export the XMODE line, in a way that is possible use separate ROM/RAM areas for each thread. In this case each thread can work as a fully separate cores, in a way that one thread known nothing about other thread, since there are no shared memory or registers between them and only the time is shared. In the case of applications where are idle most of time, this approach enables easily host 8 or 16 separate preemptive threads with no much logic, minimal overhead in the thread swap and the possibility of separate or shared memory resources.

Of course, this does not means that the interrupt handling is better done by the SMT concept, but at least is faster (as far as I can remember, around 20 clocks to blink the LED via the thread 1) when compared with the traditional approach, where the interrupt handler must save all registers before do anything and must restore the saved registers before return to the previous task. Just take a look in this example, although not so standard, it illustrates the overheads of the traditional appr [... cut ...] application, but some message based RTOs create a big overhead swapping contexts and the SMT concept, in theory, can help increase the performance.

Of course, the advantage of the traditional interrupt handler is the flexibility: instead of 8 or 16 pre-defined threads, it is possible run an arbitrary number between 1 and n. By this way I think both approaches have their pros and cons according to the application, in a way that I think is better support both ways in the future.

Regards,
Marcelo

thomascp, 2019-07-25:
Hi Marcelo,

Thank you for this detailed explanation!

Correct me if I am wrong. 
So you are making a multi-threading CPU and it is the hardware who decide and do the thread switching jobs.  And talk about different stacks, I guess you don't want an OS to run on darkriscv now, what you prefer is to use compiler and boot codes to initialize the threading context which including text, variables, and stacks. This is cool! But, have you thought about running an OS on darkriscv? Please let me know if you have any ideas about how to. Thanks!

I was planning to enlarge the memory space and adding some necessary components(for example interrupt controller and CSRs), and then to run an OS on darkriscv. But it seems that you have other thoughts.

By the way, I think "temporal multithreading" in https://en.wikipedia.org/wiki/Multithreading_(computer_architecture) will be more precise than SMT, do you agree?
I strongly suggest creating a group for darkriscv fans. Thanks!

BR,
Peng

samsoniuk, 2019-07-26:
Hi Peng,

Thank you for pointing that! You are right about it and I guess the current darkriscv implementation is better described as "coarse grained multithreaded". Keep in mind that until some weeks ago I was pretty sure that "SMT" was just a type of "SMP" 😆

About the OS, it depends... the stuff included in the darkriscv github is just a reference in order to test the core and some experimental stuff, such as the multithreading support and the cache controllers. Although most real world applications for the darkriscv are currently single threaded event loops handling internal stuff in FPGAs, I have some plans to port a proprietary RTOS, which is currently running in the 68000, to the darkriscv in the future! Anyway, as you mentioned, a large memory space will be required to run lots of concurrent tasks, probably with support for external SDRAM, a better and large cache controller, interrupt controller, ethernet controller, buffered UART  and other cool stuff.

In the software side, I think it is possible implement a small boot loader in the blockram, in a way that the darkriscv starts running the code in the blockram after the FPGA startup and the code works as a level 1 boot, loading a level 2 boot from the UART or from the SPI FLASH into the SDRAM, in a way that is possible support a more complex boot environment and boot a more complex OS from other locations, such as from a filesystem in the SPI FLASH or from a TFTP server via Ethernet.

In the case of the interrupt controller, I am not sure what you are planing, but maybe is possible implement something not so complex... my suggestion is use something like this:

- 32-bit CSR register with the interrupt handler vector
- 32-bit CSR register with the saved PC 
- 32-bit CSR register with the interrupt mask 
- 32-bit CSR register with the interrupt status
- 1-bit CSR register with the interrupt enable/disable

When an interrupt is generated, it will be stored in the interrupt status register and, case the interrupt is not masked in the interrupt mask register, the current PC will be saved in a CSR register and replaced by a different value from the another CSR register, pointing to the interrupt handler routing. In the interrupt handler routing some internal registers are saved in the stack, the interrupt status register is readback, in order to knows about the pending interrupts, all interrupts are processed, the saved registers are restored and an special "iret" instruction is executed, restoring the saved PC from the CSR register. Of course, the interrupt status and mask registers depends of the number of interrupts foreseen in the system.

Maybe it is required provide a quick way to enable and disable all interrupts at the same time, in a way that the OS working in a critical variable can prevent interrupts, avoiding race conditions case the interrupt needs work in the same critical variable. Also, part of interrupts can be "internal", such as programmable break points, unmapped instructions, unaligned instructions, etc. in a way that is possible trap some error conditions and provide extra debug features.

Regards,
Marcelo

thomascp, 2019-07-26:
Hi Marcelo,

'I have some plans to port a proprietary RTOS, which is currently running in the 68000'
Why not do it to an open-sourced RTOS?
Are there any plans to implement the interrupt controller and CSRs?
Thanks!

BR,
Peng

samsoniuk, 2019-07-26:
Hi Peng,

Basically because there are too much code already written for the proprietary RTOS and I have no resources to port and test the entire code... the original code was written mainly for the 68000 and I ported the code more than one decade ago to the Coldfire, which is almost as the same as the 68000... as long I known the darkriscv better than the 68000 and Coldfire, I think the port will be easy. Of course, in order to make the darkriscv more compatible with this code, I will need some similar features, including the interrupt controller, programmable breakpoints, UART interface for debug, external memory support, etc.

By this way, for sure I plan support the interrupt controller via some CSRs registers (see the last post about the special CSR registers regarding the interrupt handling).

Regards,
Marcelo

samsoniuk, 2019-08-01:
I will keep this issue open in order to remember to implement the interrupt controller and the CSR registers support!

samsoniuk, 2019-09-01:
Special registers to be implemented in order to support interrupts in a more basic way:

- mstatus w/ MPIE bit
- mcause w/ interrupt cause
- mepc w/ saved PC

Reference: [FreeRTOS RISCV support](https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS/Source/portable/GCC/RISC-V/portASM.S)

samsoniuk, 2020-10-22:
I am testing a prototype with the following implementation:

[image]

The gcc code is something like this:

--

But there are some details regarding the pipeline to fix.

samsoniuk, 2021-11-06:
...and after an epic battle that lasted 1 night and 1/2 of a day, we finally have real interrupt support in the darkriscv! \o/

four new CSR registers:

- mtvec with the interrupt handler
- mepc with exception PC+4
- mip with interrupt pending 
- mie with interrupt enable

three new instructions:

- csrw to write CSR registers
- csrr to read CSR registers
- mret to return from interrupt handler

works only with the non-threaded 3-stage pipeline!

https://github.com/darklife/darkriscv/commit/159eb623cc47249cd6fa066571499dfdb9da3bcc

closed 2021-11-06 by samsoniuk