darklife/darkriscv #20: big-endian support working on simulation only

samsoniuk, 2019-08-19:
The big-endian support works only in the simulation. The problem is probably related to the gcc : even in the simulation the -Os optimization appears to not match with the equivalent optimization in little-endian version of gcc.

samsoniuk, 2019-08-19:
Fixed! The problem was the BITS_BIG_ENDIAN set to 1 in the gcc. When comparing the code of putchar() for example, there is a test variable&1, where an "addi" is generated by the little-endian compiler, but a "slli" is generated by the big-endian compiler. By setting the BITS_BIG_ENDIAN to 0, the code makes sense again and works in both FPGA and simulation. This fix confirms that the DarkRISCV is accidentally "bi-endian", i.e. the design accidentally provides a way that the hardware and software works with both little and big-endian.

The affected file in the gcc:

    ./gcc/gcc/config/riscv/riscv.h
    89,91c89,91
    < #define BITS_BIG_ENDIAN 0

samsoniuk, 2019-08-20:
Of course, the problem never is so easy to solve...

samsoniuk, 2019-08-26:
After some effort to make the gcc generate big-endian output for RISCV, I found a mixed result: the .data* segment was fully generated as big-endian (there are some extra changes in binutils in order to make it work), but the .text* segment is not fully generated. Anyway, after some extra research I found that *is possible* implement a more intelligent way to handle both big end little-endian memories in the same core. This means that is possible put the compiled .data and .text in little-endian memory areas and put network frames in big-endian memory areas, in a way that the endian handling between that areas can be optimized (optimized means optimized, not transparent).

zeldin, 2021-06-29:
Hi.  Just a quick heads-up:  GCC 11 (and binutils 2.36) fully support the `-mbig-endian` command line option to generate code for big-endian RISC-V.

samsoniuk, 2021-06-30:
Wow! This is a very good news! I will take a look as far as possible! :)

samsoniuk, 2025-03-07:
Although the DarkRISCV is running as little-endian by default, I tested the big-endian setup on lots of different projects and it worked well! Of course, some changes are needed in order to correct map the generated image on the memory and on the core buses, but it works well, including the setup w/ L1 caches, SDRAM controller and external loadable binaries from SPI memory. In the future, maybe, we can include the possibility of select between little and big-endian but for now is enough knows it can work on both endians!

closed 2025-03-07 by samsoniuk

samsoniuk, 2025-05-21:
an additional issue regarding the big-endian support was finally closed and the big-endian support can be finally enabled on config.vh and config.mk!