darklife/darkriscv #23: Development tool install error

kunjalan, 2019-09-29:
Hi,
I'm having trouble development tool installing.
I followed the install script order but the following error message appears.

--------------- Error Message --------------------------------------------
/tmp/CC455VaP.s: Assembler messages:
/tmp/CC455VaP.s:437: Error: non-constant .uleb128 is not supported
/tmp/CC455VaP.s:438: Error: non-constant .uleb128 is not supported
Makefile:500: recipe for target 'all -target '_negdi2.o' failed
........
-----------------------------------------------------------------------------

How can I solve the problem?
Thank you.

from Chris Lee

samsoniuk, 2019-09-30:
Hi Chris,

The problem regards to the build of libgcc and, in fact, I found a similar problem some weeks ago with libgcc, although I am not sure that is exactly the same problem. Anyway, as long the DarkRISCV does not currently support large memory configurations, there is no much ways to use libgcc and the current firmware does not use this library. 

This means that the error can be ignored i.e. use "make; make install" and fully ignore any error output in order to build the compiler and the binutils, skipping the build and installation of libgcc.

Alternatively, case you will need use the libgcc for applications other than the DarkRISCV, you will probably get a similar version (gcc version 9.0.0 20180818) or use a different version with a different build procedure (as long the gcc version changes, the way to build changes too).

Best regards,
Marcelo

samsoniuk, 2020-01-05:
Link for my current toolchain:

bin = https://drive.google.com/open?id=1w_49ypSl2SLt2V9kRmVlPn-1VSHLpZ3k
src = https://drive.google.com/open?id=1GwEpw7nsc57Y6sP_s43DR9HuBUdvELFN

closed 2020-01-05 by samsoniuk