darklife/darkriscv #27: "srai" instruction doesn't work on ISE/Spartan 6

yur7aev, 2020-06-18:
for some reason the ">>>" operator
```
    FCT7[5] ? U1REG>>>U2REGX[4:0]) : U1REG>>U2REGX[4:0];
```
is synthesized into logic shift right by ISE 14.7. replacing U1REG with S1REG didn't help. i was able to work around this by
```
wire signed [31:0] RMDATA = 
     ...
    FCT7[5] ? $signed(S1REG>>>U2REGX[4:0])) : U1REG>>U2REGX[4:0];
```

samsoniuk, 2020-06-18:
Wow! Thank you for pointing the problem, I will make some tests here!

samsoniuk, 2020-06-19:
Well, well... it is really a very weird problem! I tested some additional changes and I get no results. 

Anyway, although I am not sure how the $signed() works, it appears to solve the problem! For some reason, we had a workaround for ModelSim, but not for other tools. Maybe with this fix, the problem is now solved in a more general way.

Thank you again @yur7aev, I will update the code with this workaround!

closed 2020-06-19 by samsoniuk

yur7aev, 2020-06-19:
it is weird indeed. i don't understand either how signed arithmetic is handled by the ise. or it's simply broken. i've tried to use your modelsim fix - and it's even worse giving constant 0 result. putting the shift into a separate module like
```
module sar(input signed [31:0] a, input [4:0] b, output signed [31:0] r)
    assign r = a >>> b;
endmodule
```
works, but that $signed voodoo seemed a bit less of an evil...

samsoniuk, 2020-06-19:
I tested some combinations, like declare both source and destination wires as signed, but I guess the problem is caused by the long decision chain, which is composed by mixed signed and unsigned results. As far as I understand, the $signal() does not add any extra logic in the circuit, but instead works like a cast that enforces that the output value is signed. As long the input value is also signed, the shift will work as expected.