darklife/darkriscv #34: Problem with sample firmware

monideepbora, 2021-01-27:
Dear author(s),

I tried to do a fresh compilation by removing all the .o and .s files from the src directory but no other changes. The compilation is successful but the simulation does not run (blank output, no banner etc). The error seems to be with one particular file io.s. 

Everything works if I use the io.s supplied in the repository but nothing woks if I compile io.s from io.c locally. I have done a meld compare for the io.s from the repo and the one that I have compiled.  The error seems to be with the location and type of the "io" symbol but I cant figure out exactly whats wrong. 

My aim is to run the riscv-tests on the core keeping the existing firmware worflow (using darkuart.v for console output in simulation).

Using gcc version 10.2.0 (GCC)

Please let me know if you can share some insight on what could be done.

Thanks,
Monideep

kingsumos, 2021-01-27:
Try adding the "-fcommon" option in the Makefile CCFLAGS

kingsumos, 2021-01-27:
More info at https://github.com/kingsumos/darkriscv/commit/54e2527be4c14e954da2519baf002ca57832adc0
Please let me know if this fix works for you so I can propose a merge request

monideepbora, 2021-01-27:
@kingsumos  The "-fcommon" flag seems to fix the problem. 

However, in your fork, the version of the codebase seems to be little older than the current HEAD at the original repository. The Makefile in the src folder contains a lot more options in the current HEAD at master. Please let me know if my observations are correct. 

Thanks,
Monideep

samsoniuk, 2021-01-27:
Hi Colleagues,

I tested the -fcommon in my environment (GCC 9.0.0) and found no problems. Another possibility is that the gp register (global pointer) is not initialized to the correct area. Anyway, I will include the -fcommon and some improvements in the linker script in the future updates.

Thanks,
Marcelo

kingsumos, 2021-01-27:
Ty @samsoniuk !
@monideepbora I have rebased the repo, check https://github.com/kingsumos/darkriscv/commit/ba5a2874aa0cf9695921f3350bb6c529740b2b69

samsoniuk, 2021-01-27:
The update with the -fcommon is available, but the other improvements in the linker script will need additional tests and will be released further! :)

samsoniuk, 2021-02-01:
I included lots of changes regarding the linker script and the boot code (ported from C to asm), resulting in a better control regarding the global pointer and stack pointer. I hope the changes are compatible with GCC 10+...

closed 2023-10-13 by samsoniuk