darklife/darkriscv #46: Why set two general-purpose register groups

vgegok, 2022-05-16:
Why are two general-purpose register groups designed in darkriscv.v,

Not like one of the riscv specifications,

Is it to support multi-core?

samsoniuk, 2022-05-16:
Thank you! I hope the darkriscv will help your research! :)

I have a pending task regarding turn the syntax more clear: one possibility is parse the ifdefs in a way is possible produce a Verilog output without ifdefs, which is far clearer. Another possibility is create separate files with separate cores (2-stage pipeline, 3-stage pipeline, 3-stage w/ threads, etc)... but I am not sure about the best approach.

Regarding the simulation, I think there is no problem, since X values does not really change anything in the physical design (the synthesis probably ignore it). One tip is try the synth with and without the change, to check if anything changes...but the X vallues are probably just ignored and results in no impact in the design.

Anyway, I did not like much about X values in the simulation, because they typically does appears when there is something really undefined or in conflict.

Finally, I use the MAC instruction mainly for DSP applications (16-bit audio processing), but it is not part of any official specification and it is intended to be encoded by hand. The format is the same as other R-type instructions, except by that the instruction field is fixed to 7'b1111111, as observed in the mac() function available in the stdio.c:

```
int mac(int acc,short x,short y)
{
#ifdef __RISCV__
    __asm__(".word 0x00c5857F"); // mac a0,a1,a2
    // "template"
    //acc += (x^y);
#else
    acc+=x*y;
#endif
    return acc;
}

```

The compiled code will be something like this, because the tools does not recognize the instruction pattern:

```
00000b2c <mac>:
 b2c:   857f                    0x857f
 b2e:   00c5                    addi    ra,ra,17
 b30:   00008067            ret

```

When called this way, the total time for a MAC will be 7 clocks, since the call/return pair will take 2 clock for call/return, plus 4 clocks regarding the pipeline flush (the darkriscv makes a prefetch of 2 instructions ahead). 

It is far faster than the soft-implemented multiply, but a better result can be reached by encoding the instruction inline by hand, in a way that complex audio filters encoded by hand can reach 1 MAC/clock.

vgegok, 2022-05-16:
I see, It is Wonderful!!!
I have completed the migration on the minispartan6 plus development board and will submit the merger in the near future. After that, I will have more learning tasks, such as coremark test, adding GPIO, SPI, HDMI, ADUIO and other functions.

I like your project very much, including your code style. When I am interested in risc-v, I can also learn a lot about hardware design and computer architecture.

Thank you very much！

vgegok, 2022-05-16:
Waiting for your processing of the general register part of the code, and I will close this issues.

closed 2022-05-16 by vgegok

samsoniuk, 2022-05-17:
I made the test here and, although it affects the Artix-7 maximum speed, I liked the results for Kintx-7, so I will change according to your suggestion:

```
                     manual            automatic
kintex-7        238MHz  1953LUT     244MHz  1806LUT (wow!)
artix-7         189MHz  1922LUT     172MHz  1732LUT
spartan-6       96MHz   1922LUT      95MHz  1909LUT
spartan-3       55MHz   2212LUT      56MHz  2165LUT

```

vgegok, 2022-05-20:
## Some warnings:

···
VCD info: dumpfile darksocv.vcd opened for output.
VCD warning: array word darksimv.soc0.core0.REGS[0] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[1] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[2] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[3] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[4] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[5] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[6] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[7] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[8] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[9] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[10] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[11] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[12] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[13] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[14] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[15] will conflict with an escaped identifier.
··

## ALL Log：
```root@vgegok-Xilinx:/home/vgegok/darkriscv# make
make -C src all
make[1]: Entering directory '/home/vgegok/darkriscv/src'
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32e -mabi=ilp32e -D__RISCV__ -DBUILD="\"Fri, 20 May 2022 16:20:08 +0800\"" -DARCH="\"rv32e\"" -mcmodel=medany -mexplicit-relocs  -S main.c -o main.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32e -c main.s -o main.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32e -mabi=ilp32e -D__RISCV__ -DBUILD="\"Fri, 20 May 2022 16:20:08 +0800\"" -DARCH="\"rv32e\"" -mcmodel=medany -mexplicit-relocs  -S stdio.c -o stdio.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32e -c stdio.s -o stdio.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32e -mabi=ilp32e -D__RISCV__ -DBUILD="\"Fri, 20 May 2022 16:20:08 +0800\"" -DARCH="\"rv32e\"" -mcmodel=medany -mexplicit-relocs  -S io.c -o io.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32e -c io.s -o io.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32e -mabi=ilp32e -D__RISCV__ -DBUILD="\"Fri, 20 May 2022 16:20:08 +0800\"" -DARCH="\"rv32e\"" -mcmodel=medany -mexplicit-relocs  -S banner.c -o banner.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32e -c banner.s -o banner.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32e -c boot.s -o boot.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-cpp -P  darksocv.ld.src darksocv.ld
/opt/riscv32-gcc/bin/riscv32-unknown-elf-ld -Tdarksocv.ld -Map=darksocv.map -m elf32lriscv  main.o stdio.o io.o banner.o boot.o -o darksocv.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-objdump -d darksocv.o > darksocv.lst
/opt/riscv32-gcc/bin/riscv32-unknown-elf-objcopy -O binary  darksocv.o darksocv.bin
hexdump -ve '1/4 "%08x\n"' darksocv.bin > darksocv.mem
#xxd -p -c 4 -g 4 darksocv.o > darksocv.mem
rm darksocv.bin
wc -l darksocv.mem
1877 darksocv.mem
mem ok.
sources ok.
make[1]: Leaving directory '/home/vgegok/darkriscv/src'
make -C sim all
make[1]: Entering directory '/home/vgegok/darkriscv/sim'
iverilog -Wall -I ../rtl -o darksocv darksimv.v ../rtl/darksocv.v ../rtl/darkuart.v ../rtl/darkriscv.v
darksimv.v:36: warning: timescale for darksimv inherited from another file.
./../rtl/config.vh:31: ...: The inherited timescale is here.
../rtl/darksocv.v:34: warning: timescale for darksocv inherited from another file.
./../rtl/config.vh:31: ...: The inherited timescale is here.
../rtl/darksocv.v:716: warning: implicit definition of wire 'FINISH_REQ'.
../rtl/darksocv.v:768: warning: implicit definition of wire 'IDLE'.
../rtl/darkuart.v:74: warning: timescale for darkuart inherited from another file.
./../rtl/config.vh:31: ...: The inherited timescale is here.
../rtl/darkriscv.v:55: warning: timescale for darkriscv inherited from another file.
./../rtl/config.vh:31: ...: The inherited timescale is here.
./darksocv
WARNING: ../rtl/darksocv.v:250: $readmemh(../src/darksocv.mem): Not enough words in the file for the requested range [0:2047].
VCD info: dumpfile darksocv.vcd opened for output.
VCD warning: array word darksimv.soc0.core0.REGS[0] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[1] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[2] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[3] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[4] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[5] will conflict with an escaped identifier.
VCD warning: array word darksimv.soc0.core0.REGS[6] will conflict with an escaped identifier.
[... 32 lines cut ...]

[... cut ...]
****************************************************************************
DarkRISCV Pipeline Report (41941 clocks):
core0: 70% running, 7% waiting (0% i-bus, 7% d-bus/rd, 0% d-bus/wr), 23% idle
****************************************************************************
simulation ok.
make[1]: Leaving directory '/home/vgegok/darkriscv/sim'
make -C boards all
make[1]: Entering directory '/home/vgegok/darkriscv/boards'
no board selected to build, done.
make[1]: Leaving directory '/home/vgegok/darkriscv/boards'
root@vgegok-Xilinx:/home/vgegok/darkriscv# 
```

samsoniuk, 2022-05-21:
Some warnings are related to an old problem that is not possible show arrays in the gtkwave, but the problem was in fact related to iverilog. As long I learned how show the arrays (in special de register set), the warnings started... just in case, I tried find a way to disable the warnings and found how enable all them, so I hope I will clean all warnings in the future! :)

vgegok, 2022-05-21:
I found a new probem:
config.vh:
 //`define __RV32E__
`define __HARVARD__

```
vgegok@vgegok-Xilinx:~/darkriscv-vgegok$ make install BOARD=scarab_minispartan6-plus_lx9 CROSS=riscv32-unknown-elf CCPATH=/opt/riscv32-gcc/bin ARCH=rv32i HARVARD=1
make -C src all
make[1]: Entering directory '/home/vgegok/darkriscv-vgegok/src'
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32i -mabi=ilp32e -D__RISCV__ -DBUILD="\"Sat, 21 May 2022 17:41:12 +0800\"" -DARCH="\"rv32i\"" -mcmodel=medany -mexplicit-relocs  -S main.c -o main.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32i -c main.s -o main.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32i -mabi=ilp32e -D__RISCV__ -DBUILD="\"Sat, 21 May 2022 17:41:12 +0800\"" -DARCH="\"rv32i\"" -mcmodel=medany -mexplicit-relocs  -S stdio.c -o stdio.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32i -c stdio.s -o stdio.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32i -mabi=ilp32e -D__RISCV__ -DBUILD="\"Sat, 21 May 2022 17:41:12 +0800\"" -DARCH="\"rv32i\"" -mcmodel=medany -mexplicit-relocs  -S io.c -o io.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32i -c io.s -o io.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-gcc -Wall -fcommon -ffreestanding -I./include -O2 -march=rv32i -mabi=ilp32e -D__RISCV__ -DBUILD="\"Sat, 21 May 2022 17:41:12 +0800\"" -DARCH="\"rv32i\"" -mcmodel=medany -mexplicit-relocs  -S banner.c -o banner.s
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32i -c banner.s -o banner.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-as -march=rv32i -c boot.s -o boot.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-cpp -P  -DHARVARD=1 darksocv.ld.src darksocv.ld
/opt/riscv32-gcc/bin/riscv32-unknown-elf-ld -Tdarksocv.ld -Map=darksocv.map -m elf32lriscv  main.o stdio.o io.o banner.o boot.o -o darksocv.o
/opt/riscv32-gcc/bin/riscv32-unknown-elf-ld: darksocv.o section `.text.startup' will not fit in region `ROM'
/opt/riscv32-gcc/bin/riscv32-unknown-elf-ld: section .data LMA [0000000000001000,0000000000001677] overlaps section .text.startup LMA [0000000000000fa8,00000000000016d3]
/opt/riscv32-gcc/bin/riscv32-unknown-elf-ld: region `ROM' overflowed by 1748 bytes
Makefile:113: recipe for target 'darksocv.o' failed
make[1]: *** [darksocv.o] Error 1
make[1]: Leaving directory '/home/vgegok/darkriscv-vgegok/src'
Makefile:50: recipe for target 'install' failed
make: *** [install] Error 2

```

Does RV32I not support HARVARD?

vgegok, 2022-05-21:
I Try again, I must to `make`, Than `make install`.

vgegok, 2022-05-21:
It is Working!!! 
👇👇👇👇👇👇👇👇👇👇👇👇👇👇👇👇👇👇👇👇
darksoc.ld.src:
```
MEMORY
{
    IO  (rw!x) : ORIGIN = 0x80000000, LENGTH = 0x10
#if HARVARD
    ROM (x!rw) : ORIGIN = 0x00000000, LENGTH = 0x2000
    RAM (rw!x) : ORIGIN = 0x00002000, LENGTH = 0x2000
#else
    MEM (rwx)  : ORIGIN = 0x00000000, LENGTH = 0x4000
#endif
```
darksocv.v:
``` verilog
`ifdef __HARVARD__

    reg [31:0] ROM [0:2047]; // ro memory
    reg [31:0] RAM [0:2047]; // rw memory

    // memory initialization

    integer i;
    initial
    begin
        for(i=0;i!=2048;i=i+1)
        begin        
            ROM[i] = 32'd0;
            RAM[i] = 32'd0;
        end

        // workaround for vivado: no path in simulation and .mem extension
        
`ifdef XILINX_SIMULATOR
        $readmemh("darksocv.rom.mem",ROM);        
        $readmemh("darksocv.ram.mem",RAM);
`else
        $readmemh("../src/darksocv.rom.mem",ROM);        
        $readmemh("../src/darksocv.ram.mem",RAM);
`endif        
    end
```
`ROM[IADDR[11:2]];`--->`ROM[IADDR[12:2]];`
`RAM[DADDR[11:2]];`--->`RAM[DADDR[12:2]];`

samsoniuk, 2022-05-22:
The core is always working with harvard architecture, but it is possible build the soc w/ a pure harvard architecture (separate ROM/RAM memories w/ the option HARVARD active) or build the soc w/ a hybrid harvard/von neumann architecture. This is done by connecting the instruction and data buses in different buses of the same BRAM, in a way that the instruction and data buses can work in parallel, as defined by the harvard architecture, but it has a unified address space, as defined by the von neumann architecture.

As long the firmware is composed by ~5KB of code and ~1.5KB of data, the hybrid memory model can easily fit both code and data in less than 8KB. However, in the pure harvard memory model we must create a 8KB memory for code and at least 2KB memory for data... to make it more simple, I just defined the same size for both, in a way that a pair of 8KB+8KB memories are created, via the option MLEN in the config.vh.

In the case of the firmware, there is a HARVARD option in the Makefile but unfortunately there is no much integration with the hardware options, so the option must be enabled by hand. Although is possible automate this integration (by parsing the config.vh in the Maketile), I am not much sure that the pure harvard model have any advantage over the hybrid model, so I will keep the hybrid memory model as default.

oh, I found a small bug in the src/Makefile, which explain why the make does not worked directly. I also cleaned the simulation warnings!

vgegok, 2022-05-22:
I saw your modification. It's great!
Looks like you're still working at night——dark:)

vgegok, 2022-05-26:
> I made the test here and, although it affects the Artix-7 maximum speed, I liked the results for Kintx-7, so I will change according to your suggestion:
>
> ```
>                      manual [...]

**I wonder how you got these scores? CoreMark？ Can you share the source code of your application?**

samsoniuk, 2022-05-26:
Those results are from the Xilinx ISE 14.7 synthesis report:

```
$ egrep "(LUTs|Frequency|Target)" darksocv.syr 
---- Target Parameters
Target Device                      : xc7k420t-2-ffg901
---- Target Options
 Number of Slice LUTs:                 1633  out of  260600     0%  
   Minimum period: 4.197ns (Maximum Frequency: 238.276MHz)

```

So, the above results are the frequency and area used in the FPGA (I just get a random report here, not sure it is the latest version in the github).

The effective performance in MIPS depends of the application: in theory it is possible peaks an IPC = 1 (instruction per clock), but it requires optimize the code by hand in order to unroll all loops and minimize the use of load/store instructions and/or use LUTRAM for data, in order to avoid wait-states (the BRAM requires 1 wait-state in the load instruction).

The default test firmware, compiled with GCC -O2, prints a detailed pipeline report at the end of simulation:

```
****************************************************************************
DarkRISCV Pipeline Report (45485 clocks):
core0: 71% run, 6% wait (0% i-bus, 6% d-bus/rd, 0% d-bus/wr), 23% idle
****************************************************************************

```
In the report is possible see who much clocks the simulation used, who much of them are lost to wait-states and pipeline flushes. In the above case, we lost 23% of the clocks to pipeline flushes, 6% to wait-states and the result is that the processor really execute useful instructions only 71% of time, which results in a IPC = 0.71. Supposing the best frequency score of 244MHz, we have 244 x 0.71 = 173MIPS.

Although the this is not a real benchmark, it is a real application, so the empirical result is good enough to check how the core is spending the clocks in the pipeline.

vgegok, 2022-05-29:
I made some updates:
1. Add io.timeus used to provide system time, similar to `clock_ t clock()`
2. Migrate coremark
3. Independent allocation of `MLEN` in Harvard and von Neumann architecture
4. modify the readme and some makefile contents
5. ........

You can evaluate whether the pull request can be submitted.

samsoniuk, 2022-05-29:
wow! I will open a new issue regarding the coremark!

samsoniuk, 2022-05-30:
I opened an issue #52, so new comments, suggestions and general posts can start there and moved to different issues as long they switches from suggestions to real implementation.