darklife/darkriscv #67: Enabling __INTERACTIVE__ option when running make file

jones6a9, 2023-11-01:
Hello,

I am using darkriscv in a project for school and am in the process of writing software testbenches. I would like to be able to include scanf functions in the main.c file, but it does not seem to be working. Is there a setting somewhere I could change to enable the interactive option/allow scanf function to be used?

Thank you

samsoniuk, 2023-11-01:
Unfortunately it was designed only to run w/ darkshell code: it prints a prompt "> , which triggers the simulation to stop on darkuart.v and wait for an input string, in a way that, you press enter, the simulation restarts and, eventually, the line is read char by char and processed by the gets() function. 

A possible workaround is try use sscanf(), in a way that you print a prompt "> ", get a string w/ gets() and process this string with sscanf(). Of course, darklibc does not include scanf() or sscanf(), so you need provide your own code. 

Eventually, I can in the future try re-code the darkuart.v to avoid the requirement for a prompt (the prompt is used to control some state machines on RX path of UART) and enable a more simplistic handshaking based only on getchar(), but I am very busy at the moment, so the workaround w/ sscanf() is probably the better way at the moment.

samsoniuk, 2023-11-01:
ohh, a very important question: why you need use the interactive mode on your simulation? note that when the prompt is waiting for stdin input, the simulation is fully stopped because the stdin read operation is blocking the execution, so the use of interactive mode is very specific, that is why it is working only w/ gets() on darkshell.

closed 2025-03-07 by samsoniuk