darklife/darkriscv #93: darkbridge.v undocumented

MrJake222, 2025-04-25:
The `rtl/README.md` doesn't mention what and why `darkbridge.v` does.

samsoniuk, 2025-04-25:
yeah, I need add a better explanation about it... but, basically, the darkbridge is a kind of bus bridge between the high speed core buses (synchronous instruction and data buses), synchronous memory (including L1 caches) and the low speed main bus (a von Neummann bus with SDRAM, peripherals, etc).

samsoniuk, 2025-04-26:
I made an update on the rtl/README.md, so there is a better explanation about the RTL modules... it is far from complete, but I think separate READMEs in each directory will probably be a better way instead of increase the main README even more :)

closed 2025-05-20 by MrJake222