darklife/darkriscv #94: Add cycle and instret counters ("Zicntr" extension)

MrJake222, 2025-04-25:
Hello,

The ISA suggests it should be implemented, as it serves basic performance analysis. I need it too for Dhrystone benchmark.
I can see that clocks and instructions are being counted with `__PERFMETER__` enabled. Why not make them accessible via CSRs?

samsoniuk, 2025-04-25:
it was not added before because we had no CSRs, so we made all measurements via Verilog counters on simulation and via time measurement... but, when we added the CSRs for interrupt and debug, it would be easy to support it, since the CSR handling instructions are present now. I will check about it.

closed 2025-04-26 by samsoniuk

samsoniuk, 2025-04-26:
thanks Jake! :)

samsoniuk, 2025-04-26:
I added support for that new registers on darklibc:

long long get_mcycle(void);
long long get_minstret(void);

I also tested them and the result matches with the result calculated by the verilog code! 

thanks again!
marcelo

PR #95 (2025-04-26): Added 2 CSR counters: cycle & instructions
Fixes #94

Ad-hoc fix, cloned the counters logic to CSR section. Down there it's really entangled into simulation logic...
commit abaae6c6 (2025-04-26): Added 2 CSR counters: cycle & instructions

Fixes #94