darklife/darkriscv #104: sdram compile for mt48lc16m16a2 fails

Equinox86, 2025-07-16:
To reproduce, this issue, I have followed the instructions on the README installing iverilog and Xiliinx ISE first. 

When running **make** from the top-level directory , the compile fails with the following errors:

> equinox@xilinx:~/darkriscv$ make
> make -C sim
> make[1]: Entering directory '/home/equinox/darkriscv/sim'
> iverilog -Wall -I ../rtl -o darksocv darksimv.v ../rtl/darksocv.v ../rtl/darkbridge.v ../rtl/darkuart.v ../rtl/darkriscv.v ../rtl/darkpll.v ../rtl/darkram.v ../rtl/darkio.v ../rtl/darkcache.v ../rtl/darkmac.v ../rtl/lib/sdram/mt48lc16m16a2_ctrl.v
> ../rtl/lib/sdram/mt48lc16m16a2_ctrl.v:154: error: Unable to bind wire/reg/memory `oe' in `mt48lc16m16a2_ctrl'
> ../rtl/lib/sdram/mt48lc16m16a2_ctrl.v:156:      : A symbol with that name was declared here. Check for declaration after use.
> ../rtl/lib/sdram/mt48lc16m16a2_ctrl.v:154: error: Unable to elaborate r-value: (oe)?(dq):(16'bzzzzzzzzzzzzzzzz)
> ../rtl/lib/sdram/mt48lc16m16a2_ctrl.v:78: error: Unable to bind wire/reg/memory `CAS_LATENCY' in `mt48lc16m16a2_ctrl'
> 3 error(s) during elaboration.
> make[1]: *** [Makefile:110: darksocv] Error 3
> make[1]: Leaving directory '/home/equinox/darkriscv/sim'
> make: *** [Makefile:43: default] Error 2

samsoniuk, 2025-07-16:
hmmmm it is probably related to the icarus version, can you run a "iverilog -v" to check?

my iverilog version is:

[image]

anyway, case you already have the Xilinx tools installed, you can skip the iverilog test and try open one of the Xilinx projects directly:

./boards/avnet_microboard_lx9/darksocv.xise
./boards/xilinx_ac701_a200/darksocv.xise
./boards/qmtech_sdram_lx16/darksocv.xise
./boards/digilent_spartan3_s200/darksocv.xise
./boards/aliexpress_hpc40gbe_k420/darksocv.xise
./boards/piswords_ch34x_lx16/darksocv.xise
./boards/scarab_minispartan6-plus_lx9/darksocv.xise
./boards/papilio_duo_logicstart/darksocv.xise
./boards/piswords_rs485_lx9/darksocv.xise
./boards/qmtech_kintex7_k325/darksocv.xise

it is possible simulate inside Xilinx ISE, the iverilog is not a dependency.

Equinox86, 2025-07-16:
Thanks for the quick reply! 

Looks like I am on version 13. 0. 

I created a pull request that presents a minimal change  - no pressure, but it seemed to work and I didn't see any dependency issues. 

> equinox@xilinx:~/darkriscv$ iverilog -v
Icarus Verilog version 13.0 (devel) (s20250103-54-ge55d9454d)

Copyright (c) 2000-2024 Stephen Williams (steve@icarus.com)

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.

iverilog: no source files.

Usage: iverilog [-EiRSuvV] [-B base] [-c cmdfile|-f cmdfile]
                [-g1995|-g2001|-g2005|-g2005-sv|-g2009|-g2012] [-g<feature>]
                [-D macro[=defn]] [-I includedir] [-L moduledir]
                [-M [mode=]depfile] [-m module]
                [-N file] [-o filename] [-p flag=value]
                [-s topmodule] [-t target] [-T min|typ|max]
                [-W class] [-y dir] [-Y suf] [-l file] source_file(s)

samsoniuk, 2025-07-16:
oh, nice! it appears that I need update my version here! thanks for the contribution! :D